hex-five/multizone-secure-iot-stack #5: Send message behavior inconsistent between zone 1 & 4

donbarnetson, 2019-02-14:
If I send messages from Zone 1:
 - send 1 a - no response (ideally should locally echo a)
 - send 2 a - no response - this is OK
 - send 3 a - no response - this is OK
 - send 4 a - zone 4 prints Z4> ???? (should print character)

If I send messages from Zone 4:
 - send 1 a - no response (should echo character on zone 1)
 - send 2 a - Zone 4 prints ???? (should not print anything)
 - send 3 a - no response (ok)
 - send 4 a - no response (should local echo character)

sandro2pinto, 2019-02-15:
Partially fixed on commit [526df4e](https://github.com/hex-five/multizone-secure-iot-stack/commit/526df4ea0729fd2f4651621fbf76e69cdd8f14ec)

Now, if I send messages from Zone 1:

* send 1 a - echo a
* send 2 a - no response - this is OK
* send 3 a - no response - this is OK
* send 4 a - print a

cgarlati, 2019-02-15:
My 2 cents: messages should be defined in a common .h using mnemonic symbols and the their behavior should be consistent across all zones - whenever possible.

cgarlati, 2019-02-15:
Don, Once finalized the expected behaviour should be clearly documented in readme and/or manual

cgarlati, 2019-02-15:
And when the listener doesn't have a specif action for the incoming message it should echo it back as is - without getting into an infinite ping / pong loop.

donbarnetson, 2019-02-16:
Z1 message behavior is now correct.  Thx

Z4 is still incorrect:
* send 1 a - no response (should echo character on zone 1)
* send 2 a - Zone 4 prints ???? (should not print anything)
* send 3 a - no response (ok)
* send 4 a - no response (should local echo character)

borancar, 2019-02-20:
These is a bit of conflict in the specification. Should `send 1 p` print the character p in zone1 or should it ping zone1? Same with `send 4 p` from zone1.

donbarnetson, 2019-02-22:
Lets treat p as a special case and do a ping/pong with it, other characters should be printed in teh zone they are sent to.

In Pull Request #42, ping/pong works, but other messages between Zones 1 & 4 do not:
Z4> send 1 a (no response in Z1)
Z4> send 4 k  (no response)
Z1> send 4 a (no response in Z4)
Z1> send 1 k (no response)

cgarlati, 2019-02-22:
This seemingly innocuous requirement has led to many code iterations without producing any satisfactory / useful mean to test and demonstrate the messaging infrastructure. There are a few non obvious corner cases to be considered. Let's clearly define the requirements before investing more time & effort.

closed 2019-02-22 by cgarlati