hex-five/multizone-secure-iot-stack #7: Zone 1 exec 20410000 causes crash

donbarnetson, 2019-02-14:
issuing an exec command to Z1 flash base 20410000 address causes zone 1 to crash

sandro2pinto, 2019-02-15:
'exec' command removed - such as 'timer' it doesn't make sense in the context of FreeRTOS

cgarlati, 2019-02-15:
Somehow agree with regard to the soft timer. Disagree on exec.

borancar, 2019-02-15:
Can you describe the crash more? What I was experiencing was telnet hang, but zone1 was still running the LED through the rainbow pattern and the robot light was responding to robot on/off. The telnet hang would be a result of not properly resetting the mzmsg communication.

donbarnetson, 2019-02-16:
Agreed - yes, it was a telnet hang; but now that command is removed it is a moot point.

donbarnetson, 2019-02-20:
exec has been removed; issue is closed.

closed 2019-02-20 by donbarnetson