hex-five/multizone-secure-iot-stack #8: telnet in before system up causes telnet to hang

donbarnetson, 2019-02-14:
If you reset the system using the prog done button and telnet in rapidly, telnet will hang for exactly 30s then punch through.

borancar, 2019-02-15:
I can't reproduce that one. I did a prog and telnet immediately and recorded in wireshark a couple of times in a row. It took about 3-4 sec before I get the response punching through. Could you describe your network setup more or potentially capture the packets with wireshark?

donbarnetson, 2019-02-16:
Can reproduce in Ubuntu as well - wireshark capture attached.
[telnet_on_board_restart.zip](https://github.com/hex-five/multizone-secure-iot-stack/files/2872434/telnet_on_board_restart.zip)

borancar, 2019-02-20:
This looks like an issue due to networking from the VM. It may take longer for the packet routing to stabilize depending on what networking mode in VirtualBox is selected. Could be useful to capture with Wireshark on Windows as well, just to see what's going on with networking.

borancar, 2019-02-20:
Suggest lowering label to P3 Bug or enhancement.

cgarlati, 2019-02-20:
My Debian openssl 1.1.1 client runs in a vmware guest and I am not able to reproduce

borancar, 2019-02-20:
vmware and virtualbox would have different networking drivers in place

donbarnetson, 2019-02-22:
Does not do it in bare metal Ubuntu 18.04, thus must have been an issue of Virtualbox drivers as suspected by Boran.  Closing.

closed 2019-02-22 by donbarnetson