hex-five/multizone-secure-iot-stack #13: multizone.jar doesn't handle continous sections bigger than 64k

borancar, 2019-02-15:
### Description

multizone.jar doesn't handle continuous sections bigger than 64k, it doesn't insert an Extended Linear Address entry and the addresses wrap around, causing an overlap. 

### Steps to reproduce
1. Create a zone bigger than 64k in code
2. Adjust the multizone.cfg and the linker scripts
3. Compile successfully with `make`
4. Try to load it with `make load`

### Expected results
Everything loads correctly.

### Actual results
```
Loading section .sec1, size 0xd46 lma 0x20400000
Loading section .sec2, size 0x8160 lma 0x20410000
Loading section .sec3, size 0x10000 lma 0x20420000
Loading section .sec4, size 0x9fbe lma 0x20420000
Loading section .sec5, size 0x460 lma 0x20460000
Loading section .sec6, size 0x4e00 lma 0x20470000
Info : Padding image section 0 at 0x20400d46 with 62138 bytes
Info : Padding image section 1 at 0x20418160 with 32416 bytes
Error: Section at 0x20420000 overlaps section ending at 0x20430000
```

cgarlati, 2019-02-15:
Fixed in efd7407. Please test thorougly and eventually close.

borancar, 2019-02-15:
Does it include oneoff-sandro-irq modifications or should we retest with the slower bitstream?

cgarlati, 2019-02-15:
It does not include irq auto disable / enable. Please test the 64K fix on this one. Will re-introduce irq auto disable / enable once this issue is tested and closed.

borancar, 2019-02-15:
Confirmed working with a 192kb+ zone. Boundaries are inserted correctly.

closed 2019-02-15 by borancar