hex-five/multizone-secure-iot-stack #16: Performance regression with latest merge, sandro2pinto-freertos_update

borancar, 2019-02-15:
### Description

Latest merge f50e39830a41f78d366dcff49e00a80a1d035675 seems to have brought a performance regression for the telnet side. Sometimes the characters are printed with a jerky motion and there is a delay between repeating or deleting characters.

### Steps to reproduce

1. Open a telnet connection
2. Hold any key for repeat characters
3. Release the key

### Expected results

Key printing should stop the moment you release the key

### Actual results

There is some delay between the keys

sandro2pinto, 2019-02-15:
Consequence from fix the telnet backspace (commit [13fc267](https://github.com/hex-five/multizone-secure-iot-stack/commit/13fc267b0043bed2cc35a46d2f135107777784c6)) - we just added the same code from the the original CLI (code from Zone4).

borancar, 2019-02-15:
Yup, `git bisect` confirms it:
```
13fc267b0043bed2cc35a46d2f135107777784c6 is the first bad commit
commit 13fc267b0043bed2cc35a46d2f135107777784c6
Author: sandro2pinto <sandro2pinto@gmail.com>
Date:   Fri Feb 15 15:42:36 2019 +0000

    Fix telnet backspace

:040000 040000 9e4317f60aded36cf63cecb88cd06522874fff7f cdec92c99b98f866f97552e5aafdb765a504c560 M	zone1
```

borancar, 2019-02-15:
Possible cause is flushing zone2 with every single write. Should probably combine those into a single mzmsg_write or add the ability to control the flush when invoking mzmsg_write.

cgarlati, 2019-02-15:
As it is right now diting it's almost unusable - it drives me crazy. Was way better before.

borancar, 2019-02-19:
Fixed in https://github.com/hex-five/multizone-secure-iot-stack/pull/24. There's still room for even faster performance if needed.

closed 2019-02-22 by cgarlati