hex-five/multizone-secure-iot-stack #19: Zone1 line editing doesn't work

cgarlati, 2019-02-15:
1) Type "send robot 0" 
2) Type enter
3) Go back to previous history with arrow up
4) move left with left arrow to 'r'
5) backspace to delete "send "
>>> Screen and buffer go out of synch and an extra white space is inserted

borancar, 2019-02-15:
Simpler example:
1. Type "pmp"
2. Move to last p and press backspace
3. You get "p p"
4. Press backspace again and you get " p"
5. Type "pm" and press enter
6. You ended up executing pmp

Conclusion, sometimes backspace works like insert/overwrite space, sometimes like backspace

borancar, 2019-02-19:
Fixed in https://github.com/hex-five/multizone-secure-iot-stack/pull/24

donbarnetson, 2019-02-20:
This appears to be fixed in my tests

sandro2pinto, 2019-02-20:
Fixed in commit [158a0c2](https://github.com/hex-five/multizone-secure-iot-stack/commit/158a0c26de0dd799d7849fd7192dff58238308d7)

closed 2019-02-22 by cgarlati