hex-five/multizone-secure-iot-stack #28: Add robot syntax help in Zone1

sandro2pinto, 2019-02-19:
It would be nice to have a robot syntax help to let users know the required commands to control the robot.

For example when you type just "robot" or "robot help" you may end having the following reply:

Syntax: robot { q | a | w | s | e | d | r | f | t | g | y }
             'q' - grip close
             'a' - grip open
             'w' - wrist up
             's' - wrist down
             'e' - elbow up
             'd' - elbow down
             'r' - shoulder up
             'f' - shoulder down
             't' - base clockwise
             'g' - base counterclockwise
             'y' - light on

cgarlati, 2019-02-20:
to simplify, and given that there should be no robot command to start with, the syntax could be displayed upon USB attach - with the adition of the 0 1  >  < options.

borancar, 2019-02-20:
Sandro and Sergio have already implemented this as part of https://github.com/hex-five/multizone-secure-iot-stack/pull/35, commit https://github.com/hex-five/multizone-secure-iot-stack/pull/35/commits/ae13338e96990bf892921a64f377069b374639b6

sandro2pinto, 2019-02-20:
Yes - already implemented.

If you type "send 1 h" you will get the syntax - I will move to display once the USB is attached as well.

sandro2pinto, 2019-02-22:
Fixed in [4740de7](https://github.com/hex-five/multizone-secure-iot-stack/commit/4740de751061cffa80d887a208239f3b106a4403)

closed 2019-02-22 by sandro2pinto