hex-five/multizone-secure-iot-stack #30: No Splash screen on ZOne 1 TLS Load

donbarnetson, 2019-02-20:
Here is what a TLS 1.3 session looks like on my system - it stops at the bottom, until I press enter and get the list of commands

dbarnets@dbarnets-VirtualBox:~$ stty -icanon -echo && openssl s_client -crlf -connect 192.168.0.2:443
CONNECTED(00000003)
depth=0 C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com
verify error:num=18:self signed certificate
verify return:1
depth=0 C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com
verify return:1
---
Certificate chain
 0 s:C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com
   i:C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com
---
Server certificate
-----BEGIN CERTIFICATE-----
MIICyzCCAnGgAwIBAgIJAN7Xyww12NPjMAkGByqGSM49BAEwgcAxCzAJBgNVBAYT
AlVTMRMwEQYDVQQIDApDYWxpZm9ybmlhMRcwFQYDVQQHDA5TaWxpY29uIFZhbGxl
eTEgMB4GA1UECgwXSGV4IEZpdmUgU2VjdXJpdHksIEluYy4xIzAhBgNVBAsMGk11
bHRpWm9uZSBTZWN1cmUgSW9UIFN0YWNrMRowGAYDVQQDDBF4MzAwLmhleC1maXZl
LmNvbTEgMB4GCSqGSIb3DQEJARYRaW5mb0BoZXgtZml2ZS5jb20wHhcNMTkwMjE5
MjEzNDEyWhcNMjExMTE1MjEzNDEyWjCBwDELMAkGA1UEBhMCVVMxEzARBgNVBAgM
CkNhbGlmb3JuaWExFzAVBgNVBAcMDlNpbGljb24gVmFsbGV5MSAwHgYDVQQKDBdI
ZXggRml2ZSBTZWN1cml0eSwgSW5jLjEjMCEGA1UECwwaTXVsdGlab25lIFNlY3Vy
ZSBJb1QgU3RhY2sxGjAYBgNVBAMMEXgzMDAuaGV4LWZpdmUuY29tMSAwHgYJKoZI
hvcNAQkBFhFpbmZvQGhleC1maXZlLmNvbTBZMBMGByqGSM49AgEGCCqGSM49AwEH
A0IABPfeThRfvW9P6rSbVkQFG5o/Cl8CKjJfQOrN7gWQLsMaoLtW18/ZjW6M9eAM
/wbd9QJi1u2Z0Uyt/dhHNUmGRQajUzBRMB0GA1UdDgQWBBQcMUyE8IyFKhLVLr5Y
m5Ml5hhW+DAfBgNVHSMEGDAWgBQcMUyE8IyFKhLVLr5Ym5Ml5hhW+DAPBgNVHRMB
Af8EBTADAQH/MAkGByqGSM49BAEDSQAwRgIhAPIpED++SWRoCkFffqxwfTzuAEDK
KvJUhLQvPNkyzGlqAiEArumnURmpRV7hMGeerTqPWKlx54WgMw/vXvLBMVK9Oyo=
-----END CERTIFICATE-----
subject=C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com

issuer=C = US, ST = California, L = Silicon Valley, O = "Hex Five Security, Inc.", OU = MultiZone Secure IoT Stack, CN = x300.hex-five.com, emailAddress = info@hex-five.com

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 1193 bytes and written 723 bytes
Verification error: self signed certificate
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 18 (self signed certificate)
---

borancar, 2019-02-20:
This is the same behavior for the telnet zone. Terminating a connection and establishing a new one just brings you back to where you were, we don't have anything in zone1 to redisplay the prompt or connection, it only happens on restart.

cgarlati, 2019-02-20:
Tend to agree with Boran

donbarnetson, 2019-02-22:
Resolved.

closed 2019-02-22 by donbarnetson