hex-five/multizone-secure-iot-stack #31: No Z1 > flush on TLS Session

donbarnetson, 2019-02-20:
For the TLS session, my system does not flush the Z1 > prompt correctly - until I start typing something

restart
press enter

======================================================================
     	                 FreeRTOS Kernel V10.1.1
     	   LEDs and Robot Control with Command-line interface
======================================================================
Machine ISA   : 0x40101105 RV32 ACIMU 
Vendor        : 0x00000000 Unknown 
Architecture  : 0x00000001 
Implementation: 0x20181004 
Hart ID       : 0x00000000 
CPU clock     : 65 MHz 
 
Tasks	    Priority	  	Description
****************************************************
cliTask        	1	Command-line interface Task
robotTask      	1	Robot Control Task
ledFadeTask    	2	LED Fade Task1
IDLE           	0	Idle

Z1 > 
Commands: load store send recv yield pmp robot stats restart

borancar, 2019-02-21:
Fixed in https://github.com/hex-five/multizone-secure-iot-stack/pull/42.

donbarnetson, 2019-02-22:
Fixed

closed 2019-02-22 by donbarnetson