hex-five/multizone-secure-iot-stack #44: Zone4 Timer doesn't work anymore

cgarlati, 2019-02-22:
Zone4 Timer doesn't work anymore

donbarnetson, 2019-02-22:
It works in my tests - what aspect is not working for you?

sandro2pinto, 2019-02-22:
Working as well with me - can you please provide more details?!

cgarlati, 2019-02-22:
Can't reproduce.

cgarlati, 2019-02-22:
To reproduce: press BTN3 once. Then timer command stops working.

borancar, 2019-02-22:
BTN3 is HW wired ot AON on the E300/X300 bitstreams.

cgarlati, 2019-02-22:
So?

borancar, 2019-02-22:
It will either stop RTC or reset it. The AON provides the RTC ticks in the E300/X300.

borancar, 2019-02-22:
So, correction, it wasn't BTN3 but BTN2 that kills the timer. Trying out with BTN3 it doesn't seem to affect the timer at all.

borancar, 2019-02-22:
@cgarlati, this one looks like a candidate to investigate in the nanokernel. I see the ECALL executing, but the trap handler never gets called.

cgarlati, 2019-02-22:
@borancar, Any ideas why this happens only with BTN2 and not with BTN1 and BTN0?

borancar, 2019-02-22:
I think I broke it even with BTN1, just make sure at least 10 interrupts happen.

borancar, 2019-02-22:
And have broken it with BTN0.

borancar, 2019-02-22:
Sometimes it may need more than 10 interrupts. Just had one case with BTN0 interrupt happening 10 times but the timer was still working.

borancar, 2019-02-22:
Broken on an older version of software, the first port to the X300 but using the latest multizone.jar.

borancar, 2019-02-22:
Broken on the E31, multizone-sdk repo master but using multizone.jar from multizone-secure-iot-stack.

cgarlati, 2019-02-23:
Issue confirmed also in sandro-one-off (automatic irq enable/disable) multizone.jar

cgarlati, 2019-02-23:
WORKAROUND: demonstrate timer command before pressing buttons or reboot the board.

cgarlati, 2019-03-04:
Sandro, per our call: what is the simplest way to reproduce in hex-five/multizone-sdk?

sandro2pinto, 2019-03-04:
Very easy. Just replace the multizone.jar file from the one delivered in the multizone secure IoT stack. Compile and load and then call the timer command and start pressing several buttons at the same time and you will see at some point in time the timer will never show.

cgarlati, 2019-03-05:
Fixed in fad5a67
hexfive-kern 81a6f3d
hexfive-conf 2cfeec8

closed 2019-03-05 by cgarlati