hex-five/multizone-secure-iot-stack #47: Zone1 Telnet: stats + btn generates two irq messages

cgarlati, 2019-02-22:


cgarlati, 2019-02-22:
press simultaneosly one button and the enter of the command stats: two irq are printed on the screen. same with TLS.

borancar, 2019-02-22:
I can reproduce without using stats. I just press BTN_0, BTN_1, BTN_2, BTN_1, BTN_0 in a row. At least one will get printed twice.

cgarlati, 2019-02-22:
Confirmed on zone4 / uart too. no need to stats: just press the three button in sequence.

cgarlati, 2019-02-22:
Boran to rollup into RC4

borancar, 2019-02-22:
Fixed in https://github.com/hex-five/multizone-secure-iot-stack/pull/48, commit https://github.com/hex-five/multizone-secure-iot-stack/pull/48/commits/e7e49df587dfc1c80dd17cb07cbfb4e22d644f10

borancar, 2019-02-22:
Debounce and the light on now at 500ms.

borancar, 2019-02-22:
FYI, double interrupts reproduced on the E31, multizone-sdk master but using latest multizone.jar from multizone-secure-iot-stack.

closed 2019-02-23 by cgarlati