hex-five/multizone-secure-iot-stack #54: Bump to hexfive-lib v1.1.0

cgarlati, 2019-03-06:
Enhanced ECALL_SEND() / ECALL_RECEIVE()

@borancar remove mzmsg complexity / overhead

borancar, 2019-03-12:
Done in https://github.com/hex-five/multizone-secure-iot-stack/pull/56.

mzmsg_write and mzmsg_read still remain as there needs to be a write/read function that splits the transfer into chunks. Happy to rename if you could provide me with a better name.

Future-wise, there's a potential to abstract it behind a file descriptor with queues and implement read/write functions.

cgarlati, 2019-03-12:
Please rename msg_write() and msg_read()

borancar, 2019-03-12:
Done in https://github.com/hex-five/multizone-secure-iot-stack/pull/56/commits/6269239549a5869d09716be74f167fa4fe7ff44d.

closed 2019-03-12 by cgarlati