hex-five/multizone-secure-iot-stack #57: LEDs stop blinking if buttons pressed without a telnet connection

borancar, 2019-03-12:
### Steps to reproduce

1. Connect via telnet
2. Disconnect telnet
3. Press any button repeatedly

### Expected results

LD0 should continue blinking and LD1 should continue following the rainbow pattern

### Actual results

LD0 stops blinking and LD1 is stationery

borancar, 2019-03-12:
The issue arises because zone1 has not concept of the telnet disconnect. It believes that the telnet connection is still there based on the buffers still containing some space. When that space runs out eventually, the whole zone1 gets stuck yielding at any of these msg_writes: https://github.com/hex-five/multizone-secure-iot-stack/blob/d9ee627af409204e9baec830c0333b742a753109/zone1/cli.c#L468-L479

borancar, 2019-03-12:
Recommend passing to @sandro2pinto, as a complete solution needs to involve refactoring the FreeRTOS code and he might have a better idea how to tackle it.

cgarlati, 2019-03-12:
@sandro2pinto Send message should not block the freeRTOS task - perhaps you need a message from the tcp/ip zone signaling the lost connection so that you can clear the buffer. Or you could introduce a timeout.

closed 2019-03-27 by cgarlati