hex-five/multizone-secure-iot-stack #60: Support a mechanism which facilitate one zone calling another zone's service api

cui632, 2019-04-25:
Most application from user and open source packages are designed with RTOS service API, they are not willing to change code to adapt also, so it will be disaster in the case of Mutlizone can’t support the design. there are more examples which would be found in google. it is normal for user zone to call system zone's service. 
it also brings new challenge to current design. 
1.  it is not data or message transferration indeed. 
2. system zone might access user zone's paramters, but it must be fail since they are in different zone. 
3. how does user zone get result from system zone, it might access the data in reference to reach desired performance.
4. like FreeRTOS with more RTOS api which are expost to user calling, how to arrange those api in some structure to be call from external zone. it might bring code size bigger than before. 

Vincent

borancar, 2019-04-25:
Hey Vincent,

Are you asking for RPC/IPC here? At the moment, it's up to the user to code his own RPC/IPC protocol using ECALL_SEND/ECALL_RECV manually, see eccSign for example:
Here's the caller in zone2:
https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone2/main.c#L377

The selector in zone3:
https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone3/main.c#L122

And the implementation:
https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone2/main.c#L377

The serialization/deserialization can definitely be handled by using an existing RPC/IPC framework, e.g. [Avro](https://avro.apache.org/), [Protobuf](https://developers.google.com/protocol-buffers/), [Cap'n'proto](https://capnproto.org/)...

We are evaluating the security/performance tradeoffs of these protocols as well as the implications of shared memory between zones. At the current context switch time of ~3us and zone2<->zone3 switch of <200us on a 65 MHz FPGA implementation we found the performance sufficient for the IoT stack example so we didn't pursue it in the example. MultiZone will, however, let you create a shared memory region between zones if you need faster performance and we can help with choosing a right serialization/deserialization and RPC/IPC framework.

sandro2pinto, 2019-04-25:
Hi Vincent (@cui632 ),

thank you for exposing your question so that we can keep track of all outstanding issues.

The example that Boran described to you is exactly a proof of concept that is possible to implement what you are asking for on top of MultiZone. In that example, the API is not from FreeRTOS but from WolfSSL. You can apply a similar rationale to implement for FreeRTOS.  
In fact, you just need to implement a set of wrappers that implement an RPC/IPC mechanism based on the Secure InterZone communication infrastructure (i.e., ECALL_SEND/ECALL_RECV) and then apply all the marshalling/unmarshalling (serialization/deserialization) to handle the messages. You can either elaborate a bit more and in the context of FreeRTOS, you can either use the concept of proxy Tasks (i.e., server stub, client stub).

Anyway, what I'm missing in your description, is what's your definition of "system zone" and "user zone". If I'm correctly understanding your current vision, I need to add that you don't need necessarily to pick a legacy application composed by an N number of FreeRTOS Tasks and run all of these Tasks into N different Zones. As you said, in that case, you will end with much more complexity, due to the systematic interactions between these zones. This is totally against the principles of MultiZone. FreeRTOS and the several Tasks can run into a single zone such as demonstrated in the Secure IoT stack demo, and just the pieces of code which need to be necessarily isolated can be offloaded to a different zone. 

All in all, just to sum up, it is possible to implement what you are describing (as already demonstrated by the wolfSSL API), but it should not be generalized for all single applications that run in a legacy FreeRTOS application - just the ones that make real sense.  

Sandro

cui632, 2019-04-25:
This sample is a simplest example, which also need two parts work colleborately by extra code. it is not good idea. 
The following code is too tough, no designers will change own application like this. , you know what I mean ? it is only sample to all eccSign function, if appliation is complex enough, it will be crazing. 

system zone  - the zone is running with FREEROTS
user zone      - the zone is running with user application. 
-------------------------------------------------------------------
 while (1) {
        int msg[4];

        ECALL_RECV(2, msg);
        switch (msg[0]) {
            case 1:
                eccSign((unsigned int)msg[1], (unsigned int)msg[2]);
                break;

            default:
                break;
        }

        ECALL_RECV(1, msg);
        if (msg[0]) ECALL_SEND(1, msg);
        ECALL_RECV(4, msg);
        if (msg[0]) ECALL_SEND(4, msg);
        ECALL_YIELD();
    }
---------------------------------------------------------

sandro2pinto, 2019-04-25:
@cui632 
In this case I just see 2 options:

1. You run the "system" (FreeRTOS) and the "user" (Application) into a single "system+user" zone - and you don't need to change absolutely nothing.

2. You run a "system zone" (FreeRTOS) and a different "user zone" (Application) and Hex-Five provides an already implemented layer of abstraction (wrappers) for FreeRTOS (or other RTOS) that hides all this RPC/IPC complexity. Designers just need to include together with the Application (don't need to change the application) while everything seem to be working transparently.

cui632, 2019-04-25:
Consider system application reuse, we have to do that.  like below code, it must assure to forbid other zone doing same things before the zone receives all data.  it is still a bit of fucky, I think the nanokernel must limit the zone switch in the case.  am I right ?

void eccSign(unsigned int inSz, unsigned int outSz)
{
    static unsigned char *in;
    static unsigned char *out;
    int msg[4];
    int i, ret;

    in = (unsigned char*)malloc(inSz);
    out = (unsigned char*)malloc(outSz);

    i = 0;
    while (i < inSz) {
        ECALL_YIELD();
        ECALL_RECV(2, msg);
        if (msg[0] > 0) {
            memcpy(in+i, &msg[1], msg[0]);
            i += msg[0];
        }
    }

    ret = wc_ecc_sign_hash(in, inSz, out, &outSz, &rng, &eccKey);
    msg[0] = 1;
    msg[1] = ret;
    msg[2] = outSz;
    msg[3] = 0;
    ECALL_SEND(2, msg);
    ECALL_YIELD();

    i = 0;
    while (i < outSz) {
        int len;
        if (outSz - i > 12)
            len = 12;
        else
            len = outSz - i;

        msg[0] = len;
        memcpy(&msg[1], out + i, len);
        ECALL_SEND(2, msg);
        ECALL_YIELD();
        i += len;
    }

    free(in);
    free(out);
}

borancar, 2019-04-25:
Vincent, I think you have an older version there, which relied on the message being zero when there was no message, please look at the latest master code and it should be clearer. Here the ECALL_RECV returns whether there was data available or not, and reading the data pops it. In this sense, the transport is guaranteed, and not funky:

https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone3/main.c#L54-L61

cui632, 2019-04-26:
Dear BoranCare:
I pull up the latest code, and check it in detial , it  is formal wrapper only. but I want to say, the message mechanism is too tough also,  and the following code can't be allowd in our software review.  it is easy to cause a zone halt forever in some case, such EMC or other bad environment. 
=======================

    while (!ECALL_SEND(2, msg)) {
        ECALL_YIELD();
    }
======================-

As I discuss with Sandro last night,  when two zone are working in communication with message, other zones can't get control to run at that time. this might be limit by the message mechanism and work out by nanokenel.  let's put this way, if two zone are busy to send and recieve a mount of data, and it really takes  a long time. the transit must lock the 16 byte share buffer inside until unlock it at end of transit. when  other zones get control to run and also want to talk with other zones, the 16 bytes share buffer will be resused and overrided , then it enters into a disorder condition.  it brings more problem in system running in further. 

Vincent

cui632, 2019-04-26:
The latest demostration only show how to send data or message to external zone instead of call one serviice api in other zone. 
I am glad to see there is a demostrate for it .

borancar, 2019-04-26:
Ah, I see what confuses you, the 16 bytes receive buffer is not shared across all zones - each zone gets multiple mailboxes, the same as the number of zones. So in the multizone-secure-iot-stack, zone 2 has 4 mailboxes:
- from zone1
- from zone2 (itself)
- from zone3
- from zone4

When zone 3 sends to zone 2, it writes into the "from zone 3" mailbox, the nanokernel ensures this. Also, in the latest version of MultiZone, you can't overwrite the message - if the mailbox is full, ECALL_SEND will return 0 until you first ECALL_RECV the old data.

As for locking the zone forever, our example for zone2 <-> zone3 is an example of moving eccSign as a function from one zone to the other, we expect eccSign itself not to lock up. To see a more realistic scenario, where data may be stalled, have a look at zone2 <-> zone1 communications (each handling locking-up differently):

Zone1 receiving from zone2 will return the number of bytes received, and the application must handle that:

https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone1/msg.c#L12-L23

And zone2 will use queues:

https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone2/main.c#L221-L231

https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone2/main.c#L193-L199

This is one way of handling it, when you don't have tasking/threading capabilities. With tasking/threading capabilities, you can use what zone1 sending to zone2 does with portYIELDs:

https://github.com/hex-five/multizone-secure-iot-stack/blob/2eaee2423278d9a2a05d846d666c4f1e3d54de49/zone1/msg.c#L25-L54

We tried to cover as many cases as possible with our demonstration. One of these might be suited to your use-case more than others.

cui632, 2019-04-26:
Let me clarify and summrize my question.
1. the message buffer maximize lenght is 16 bytes. 
the demostrate does show a way to send data to another zone. but there is weak of it.  if one zone need to send 8K Byte data to another zone.  the zone switch would happen 8K / 16 times between those two working zones. and other zone can't get control during those time.
2. there is no demostration to show how one zone call another zone's expose API directly.

borancar, 2019-04-26:
Not necessarily, zone1 is still doing other work while sending text across thanks to the portYIELD() or the preemptive timer which just switches context to another task on FreeRTOS. Even in the case of zone2, you can still ping the board while sending text or receiving text - it's just that you need to handle multi-tasking either via a FreeRTOS or by designing your applications to be reactive, like zone2.

cui632, 2019-04-26:
But we can assure others task or zone cannot call other zone api recursive 

发自我的 iPhone

borancar, 2019-05-10:
Vincent, have a look at https://github.com/hex-five/multizone-secure-iot-stack/pull/62. This may provide answers you're looking for. Some of the points you've raised here like 8K / 16 transfers to send a message across aren't resolved, but they are decoupled by the virtue of protobuf streams that allow the underlying transport to change as needed.

closed 2019-10-18 by cgarlati