hex-five/multizone-secure-iot-stack #61: Stack Porting Issue

cui632, 2019-05-08:
Hello:

The demostration show PicoTCP and WolfSSL working with RAW API because they runs in isolation zone form others.  Bust in most use case, all stacks such as TCP, SSL/TLS must use mailbox and other service call in FreeRTOS, hereby, is there a solution about it  ? or only merge them into same zone to run ?

borancar, 2019-05-10:
Have a look at https://github.com/hex-five/multizone-secure-iot-stack/pull/62. We use protobuf to decouple the serialization/deserialization and this should allow more complex scenarios such as doing RPC from your PC to the embedded board transparently. At the moment we hardcode the function IDs, but we plan to add grpc as well.

cui632, 2019-05-11:
Does the mechanism support multi call using the protobuf together?