hex-five/multizone-secure-iot-stack #65: [Zone2] conflicting values FP_MAX_BITS / HAVE_FFDHE_4096

danielinux, 2019-09-09:
In Zone2, wolfssl user_settings.h contains `HAVE_FFDHE_4096`. This define requires `FP_MAX_BITS` to be at least 4096, or compilation fails with the following error:
`FFDHE parameters are too large for FP_MAX_BIT as set`

See [this wolfssl forum post](https://www.wolfssl.com/forums/topic1438-hexfiveerror-ffdhe-parameters-are-too-large-for-fpmaxbit-as-set.html).

Possible solutions:
- Increase  `FP_MAX_BITS` to 4096

or

- Comment out `HAVE_FFDHE_4096`

cgarlati, 2019-09-09:
@danielinux Can't reproduce.

Make sure you use Hex Five reference GNU toolchain https://hex-five.com/wp-content/uploads/riscv-gnu-toolchain-20181226.tar.xz

dgarske, 2019-09-09:
@cgarlati : Did you try with the wolfSSL submodule updated to latest master?

cgarlati, 2019-09-09:
@dgarske The stable release of the MultiZone IoT Stack is certified for wolfssl 3.15.7 @272181b. We have no plans to try different versions.

closed 2019-10-18 by cgarlati