hex-five/multizone-secure-iot-stack #66: Possibility to run HexFive IoT stack on SiFive based FreedomStudio + QEMU

coco21, 2019-09-12:
Hi team,

We already talked via wolfSSL forum and I was wondering if your nice IoT stack demo could possible be run on SiFive's provided QEMU platforms part of their SDK: FreedomStudio-4.7.2.2019-05-0-win32.win32.x86_64.

I know I can go and buy maybe a FPGA board but am seeking to make this run on (QEMU) simulation first.

Thx for your valuable help!
Cheers coco21

coco21, 2019-09-13:
Hi Casare,

Thanks for your answers. Sure going with the FPGA would be quicker, I've my reasons why I need it running on simulation - but I consider rolling with FPGA too, later on maybe.

That said I've two other questions

- What's the difference between the free MultiZone version and the commercial one? Understood it's about zones being hardcoded yet. Other things I shall be aware of? A page where such differences would be listed?
- The multizone.jar seems to be the item HexFive protects in the free version, no problem of course. In the commercial version, is the mutizone mixer program open and do we get the sources as well? Last but not least: what exact HEX format is multizone.jar outputting (I can decode but asking is faster ;))?

Thx much for your help
Chris

cgarlati, 2019-09-13:
@coco21 Hi Chris, 

> What's the difference between the free MultiZone version and the commercial one? 
The only difference is the license. Code and functionality are the same.

The commercial license entitles Hex Five's customers to:
- commercial use
- patent protection and indemnification
- clean room environment access to source code review
- hot line technical support
- early access to new features and security fixes
- discounted upgrade to new versions 
- discounted non recurring engineering work 

> Understood it's about zones being hardcoded yet.
Nothing is hardcoded or otherwise functionally constrained. Under the terms of the evaluation license you can modify zones' code and security configuration as needed

> Other things I shall be aware of?
For pricing and Pilot Program contact https://hex-five.com/contact/  

> A page where such differences would be listed?
No functional difference, so no page to list them 

> The multizone.jar seems to be the item HexFive protects in the free version
All Hex Five IP is protected domestically and internationally: patent pending US 16/450,826 PCT US19/38774 [62/689,205 62/796,967 62/796,991 62/801,091]. Except for the core technology, all Hex Five source code is published on GitHub at https://github.com/hex-five  

> In the commercial version, is the mutizone mixer program open and do we get the sources as well?
Hex Five customers are entitled to clean room review of all source code under the terms of the commercial license.

> Last but not least: what exact HEX format is multizone.jar outputting?
100% standard Intel Hex format - see https://en.wikipedia.org/wiki/Intel_HEX 

Cesare

coco21, 2019-09-13:
Hi Cesare,

Thanks much for this information, will digest all this now.
Regarding HexFive IP what are the patents exactly protecting? Sorry for asking, you must not answer it but I am wondering and tried to find some information for this one as well. If you don't mind I'd be glad, Thanks.

Have a nice week-end
Best regards,
Chris

cgarlati, 2019-10-18:
@coco21 Hi Chris,
I understand you managed to run MultiZone with the Imperas Simulator - congrats!

Can we close this issue?

And would you be so kind to leave a note here explaining how to run MultiZone with Imperas for the benefit of future readers?

Cheers
Cesare

coco21, 2019-10-18:
Hi Cesare,

I managed to setup an Imperas simulator platform close enough to the X300 FPGA platform, very close. Imperas already provide all necessary blocks and models, it was more of hacking around and recreating a X300 platform, adapt mappings and memory layouts, that's it.

What I started is also to write a special HEX loader seeing the format being used coming out of the multizone.jar tool. This one does not exist yet on Imperas side. Hex loader is implemented into harness layer for those knowing a bit about Imperas.

You can closet this ticket now I think and thanks for coming back here.

Cheers
Chris

closed 2019-10-18 by cgarlati