hex-five/multizone-secure-iot-stack #67: Running Apache MiNiFi on MultiZone Secure IoT Stack

advancedwebdeveloper, 2019-12-17:
Hi there. I was curious what are your assumptions and your vision for executing [https://github.com/apache/nifi-minifi-cpp](https://github.com/apache/nifi-minifi-cpp) . Would that be a port (hence that MiNiFi is implemented in C++) or there should be an alternative product, which would use same protocols?

That project might have various heavyweight dependencies, which themselves could be a matter of separate github issues, on your IoT stack:

libuuid
librocksdb
libcurl-openssl (If not available or desired, NSS will be used)
libssl and libcrypto from openssl
libarchive
librdkafka
libusb -- Optional, unless USB Camera support is enabled
libpng -- Optional, unless USB Camera support is enabled
libpcap -- Optional, unless ENABLE_PCAP specified
.

Ivan

cgarlati, 2019-12-17:
Hi Ivan,

Thank you for reaching out!

By design the **MultiZone Security TEE** runs independent fully linked binaries, one in each zone, and has therefore no dependencies nor requirements on the upper layers of the software stack or the programming language used to produce each individual binary. In other words, MultiZone is stack and language agnostic: it runs anything you throw at it - even complete garbage that continuously raises exceptions trying to cause DOS - that's the all point about the zero-trust model.

We have no views of preferences with regard to the any software components that you may want to run securely in any given zone. The **MultiZone Secure IOT Stack** is just an example of the many possible ways of securing a typical resource-constraint device that needs some sort of connectivity. FreeRTOS, picoTCP, wolfSSL and TLS 1.3 / ECC are best-in-class open source components and standards designed and optimized for embedded applications. They are quite good but certainly not the only options out there. 

When it comes to security, less is more - as in reduced attack surface. So my only suggestion is to stay away from thick code base software / frameworks ported to, rather than designed for, the unique challenges of embedded applications.  

Good luck with your RISC-V project!

Cesare

advancedwebdeveloper, 2020-01-06:
[https://issues.apache.org/jira/browse/MINIFICPP-1112](https://issues.apache.org/jira/browse/MINIFICPP-1112)