hex-five/multizone-secure-iot-stack #68: Supporting multizone IoT stack on the Vega Bord

advancedwebdeveloper, 2019-12-17:
Hi.

I wonder if hardware capabilities would allow this 32bit board to support your stack: [open-isa.org](open-isa.org).

Ivan

cgarlati, 2019-12-17:
Hi Ivan,

Thank you for reaching out!

**MultiZone Security** requirements for RISC-V based systems are:
1. **U-Mode** (rv32/rv64 IMACU) and
2. **PMP** (Physical Memory Protection)

If the hardware meets these requirements the MultiZone TEE should run with little no modifications - see http://hex-five.com/faq

Please note that the MultiZone Secure IOT Stack includes three additional components: **FreeRTOS**, **WolfSSL**, and **picoTCP**. You may want to check with their respective maintainers whether your hardware is compatible with these software - i.e. bare metal drivers, ram requirements, etc. 

Good luck with your RISC-V project!

Cesare

advancedwebdeveloper, 2019-12-28:
@cgarlati , I am uploading all the manuals. Could you check if U-mode and PMP are defined as implemented ones?

[ri5cy_user_manual.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007398/ri5cy_user_manual.pdf)
[zero_riscy_user_manual.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007399/zero_riscy_user_manual.pdf)

[Smart Lock on RV32M1-VEGA Board.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007400/Smart.Lock.on.RV32M1-VEGA.Board.pdf)
[RV32M1-VEGA-SCH.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007401/RV32M1-VEGA-SCH.pdf)
[RV32M1-VEGA-LAYOUT.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007402/RV32M1-VEGA-LAYOUT.pdf)
[RV32M1DS_Rev.1.1.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007403/RV32M1DS_Rev.1.1.pdf)
[RV32M1_VEGA_Quick_Start_Guide.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007404/RV32M1_VEGA_Quick_Start_Guide.pdf)
[RV32M1_Vega_Develop_Environment_Setup.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007405/RV32M1_Vega_Develop_Environment_Setup.pdf)
[RV32M1_VEGA_Board_User_Guide.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007406/RV32M1_VEGA_Board_User_Guide.pdf)
[Develop_Multicore_Application_on_RV32M1.pdf](https://github.com/hex-five/multizone-secure-iot-stack/files/4007407/Develop_Multicore_Application_on_RV32M1.pdf)

Ivan

cgarlati, 2019-12-30:
@sandro2pinto @advancedwebdeveloper Thank you Ivan for uploading the files:

**MultiZone® Security** support for **NXP RV32M1** based on tech doc:

- core #1 M0: no MPU => Not suitable for secure applications (turn off)
- core #2 M4F: yes MPU => **MultiZone® Security** Ok (G/A Feb 2020)
- core #3 ZERO-RISCY: no PMP/U-mode => Not suitable for secure applications (turn off)
- core #4 RI5CY: documentation not clear: what is the value reported by the MISA register?